uugear/Witty-Pi #1: init.d/wittypi path issues

JerksonS, 2015-08-11:
Before you copy the init.d script to /etc/init.d/wittypi  you should replace the paths used in that script.

I cloned your github while sitting at /home/Jerkson/downloads and it created the WittyPi directory.
Then, I ran the installer from /home/Jerkson/downloads/WittyPi/installWittyPi.sh
Therefore the path to the daemon.sh file would be /home/Jerkson/downloads/WittyPi/wittyPi/daemon.sh
 yet the script references /home/pi/wittyPi/daemon.sh

I have obviously updated the paths in the init.d script, but the average person might not know why the deamon isnt starting for them..

JerksonS, 2015-08-11:
Instead of being lazy I have fixed this issue in your Installer script. Please see the pull request

shawnu, 2015-08-11:
Thanks Jerkson :+1:

closed 2015-08-11 by shawnu