uugear/Witty-Pi #4: Summer time /  winter time problem

mistyk, 2015-11-03:
I think there is a big problem with the time change on summer or winter (+1 h / -1h) because when i set BEGIN at 8h the scheduler said "Next scheduled at 7h" Cf console :

WPI FILE:
BEGIN   2015-08-03 08:00:00        
END     2025-07-31 23:59:59

Result when set schedule with wittypi.sh : (Option 5)

--------------- 2015-11-03 19:54:36 ---------------
Schedule next shutdown at: 2015-11-04 17:00:00
Schedule next startup at:  2015-11-05 07:00:00
---------------------------------------------------
  Done :-)
  Please try not to shutdown your Pi manually, and let me do it for you.
  Or at least remember to turn it ON before next scheduled shutdown.

mistyk, 2015-11-03:
i've modified thant lines in runscript.sh but i don't know if it is a good  idea ...

          if [[ ${states[$index]} == ON* ]]; then
-->            setup_on_state $((check_time+3600))
          elif [[ ${states[$index]} == OFF* ]] ; then
  -->          setup_off_state $((check_time+3600))
          else

shawnu, 2015-11-04:
I think the problem is that your schedule script starts from 2015-08-03, which is in summer time, while your current date is 2015-11-03, which is in winter time. The time will be calculated from the beginning of the script, so you will lose 1 hour.

This could be avoided by setting the begin time of schedule script to any date in winter time, for example:

BEGIN 2015-11-01 08:00:00
END 2025-07-31 23:59:59

mistyk, 2015-11-04:
Yes you're right ! But i think its a issue because if I apply your parameters in WPI this issue will be already there the next summer ! do you think it's possible to fix it ?

Thanks for your job.

shawnu, 2015-11-04:
It will not be easy to fix it in Witty Pi's software, because the realtime clock doesn't take care of summer/winter time. Trying to manage summer/winter time for all countries in a script will be a challenge.

I would suggest you to write a separated program to automatically update the schedule script when summer/winter comes. It would be much easier.

mistyk, 2015-11-04:
Yes it could be a good idea, you can detect summer/winter time with this perl command in a cron job

if perl -e 'exit ((localtime)[8])' ; then
    echo winter // UPDATE TIME
else
    echo summer // UPDATE TIME
fi

Thanks.

closed 2015-11-04 by shawnu

mistyk, 2015-11-04:
I think you could keep this issue and add Enhancement label ? (because it's not really fixed.. )