uugear/Witty-Pi #5: Witty-Pi keeps shuting down at boot when connected to the internet

Tim75, 2015-12-16:
Hi, 

I am having a issue with the witty pi board.
I followed all the installation instruction but find out after the first reboot that the witty pi daemon keeps shutting down my raspberry when it is connect to the internet.
As i boot I can read " Pending for incoming shutdown command - Shutdown command received"

Disconnecting the raspi from the internet let me log into it.

As I run the wittyPi.sh I can see that the format of the system time is not qui te same as the format of the RTC time:
system time: day month year HH:MM:SS
RTC tim:       month day year HH:MM:SS

Could it be the reason of the problem?

Any suggestion on how I could fix it?

Many thanks,

shawnu, 2015-12-16:
If connecting to network can cause this problem, I guess it relates to the power supply. Maybe the network activity during the booting drawn a lot current and cause the instability of voltage, which was recognized as a shutdown command by mistake. 

If that is true, a workaround is to give more time to the GPIO pins, to reach a stable state. You can modify the "daemon.sh" script, in line 44:

while [ $counter -lt 5 ]; do

Try to change the number 5 to 25. The bigger number you use, the later the GP-4 listener will be started. This modification may workaround the problem you mentioned.

Tim75, 2015-12-16:
Many thanks for your quick reply, this has solved the problem (I am using a wifi dongle).

May I ask you another question? 
Is the witty pi board stackable with a grove pi board?

shawnu, 2015-12-16:
Grove Pi uses I2C address 0x04 while Witty Pi uses 0x68, there should not be any conflict, so I guess the answer is positive.

closed 2015-12-16 by shawnu