uugear/Witty-Pi #6: Witty Pi with Grove Pi

Tim75, 2015-12-16:
Hi, 

I am trying to use a witty pi to manage power of a raspi also connected to a grovepi board.

I am able to shutdown the raspi with wittyPi.sh, but the red light of the raspi and the green light of the grovepi both stays ON.
On the other hand I am not able to restart the raspi using the wittyPi.sh. I have also noticed that with the grovepi connected, the white LED of the WittyPi does not blink anymore after shutdown.

Is there any setting of the WittyPi I could change to make it work with a grovepi?

Thanks,

shawnu, 2015-12-17:
If all GPIO pins are used by grove pi, that will be problem. Witty Pi uses GP4 for shutdown, GP17 for lighting the LED, GP14 for detecting system shutdown. If they get used by grove pi, there will be problem.

Tim75, 2015-12-18:
Thank you Shawn for the reply.

I checked and grove pi used GPIO 0,1, 8,9,10,11,14,15 (14 and 15 are used to enable connection to a UART sensor), so the only pin posing problem is GP14 that is used both by witty pi and the grove pi. 

As I am not using any UART sensor on the grove pi could you suggest a solution so that GP14 will be used only to communicate with the Witty Pi and not the Grove pi?

Thanks,

shawnu, 2015-12-21:
I guess you will need to make some hackings on Grove Pi board. Please try to contact their support for further help.

closed 2015-12-21 by shawnu