uugear/Witty-Pi #7: WittyPi shutting down

bhozjan, 2015-12-30:
Hi,

Just received my WittyPi but it is shutting down the RPi almost immediatly after the startup.

After reading another issue, I tried to modify the daemon.sh file:
"while [ $counter -lt 5 ]; do"
by replacing 5 by 60 without correcting the issue, the RPi shutdown after 60s

I tried with 180 but without more success.

I tried without the ethernet cable without success.

Any idea to avoid the shutdown when the GPIO-4 listener is starting?

Thanks
Ben

shawnu, 2015-12-31:
Hi Ben, if delaying the starting of GPIO-4 listener doesn't help, it means your GPIO-4 was pulled down at the moment, which should not be the default state. You can start Raspberry Pi without Witty Pi and type "gpio readall" to see the state of all pins. Or you can measure the voltage on GPIO-4 with multimeter.

If you confirm GPIO-4 is in LOW state, you can check if there is any other hardware/software made this happen. Or you can workaround it by using another pin different than GPIO-4, here is a tutorial: http://www.uugear.com/portfolio/change-the-pin-that-used-by-witty-pi/

bhozjan, 2015-12-31:
Hi Shawn,

Thanks for these information, I will check these options and come back to
you (next year ;-) ).

Have a great evening!

Benoit

bhozjan, 2016-01-01:
Hi Shawn,

Let me first wish you an Happy New Year!

Here is the update:
GPIO-4 is in LOW state, I was using an SD Card with an Home Automation software pre-installed. I tried to not launch the software at boot but without any change.
So I tried a custom pin (GPIO-8) and it seems to be ok now.
I will check using a clean SD Card.

Thanks for your support.

Regarding the scheduler, is it possible to use it more as a timer than an alarm? I mean starting up every 10 minutes then shutdown after the small job done (classic logger).
Not tested but not sure that wildcards can help (?? ??:?0)

Best regards,

Benoit

shawnu, 2016-01-02:
Happy New Year Benoit :-)

I think the schedule script is exactly what you need. You can find details in the user manual, or you can follow this tutorial: http://www.uugear.com/portfolio/use-schedule-script-to-control-your-witty-pi/

bhozjan, 2016-01-03:
Hi Shawn,

My mistake, I only read the beginning of the page...

Indeep it is close to what I would like to achieve, the 5 min ON every 20 minutes is great.

To optimize the energy I plan to shutdown the RPi by code, after sending my logging data, it is less than 5 min. Of course I could modify the script "2 min ON every 20 min" for instance but 2 min could be too long or too short. My fear is that I could generate a deadlock.(I have already generated this issue ;-) )

By the way, what you provide already help me a lot, thanks for that. I'm just challenging to see if I can go further or if you have some plans for that in your roadmap.

Goal 1: Starting up every 10 min:
I can easily perform a Startup every 30 min using the "3. Set time for auto startup" with "?? ??:30" and shutdown the RPi after the end of my job.
But to extend the behavior to every 10 min, a "?0" wildcard on the minutes could work.

Goal 2: Adding exception to limit the day range
Using a script, you define a BEGIN and END date. As I don't need to log during the night I could even save energy by avoiding startups during the night.

Thanks.

Best,

Benoit

shawnu, 2016-01-03:
By using the schedule script, it is already possible to startup for every 10 minutes. It is not possible to specify "?0" wildcard for startup time, since it is not supported by the RTC chip (hardware limit, more details could be found in the datasheet of DS1337). For complex scheduling, the schedule script is the correct way to go.

If you only want to schedule startup but don't want to schedule the shutdown, you can modify the "runScript.sh" and comment out the content of setup_on_state function.

I don't understand your Goal 2. It is easy to define a schedule script that never startup at night, but the BEGIN and END date have nothing to do it.

bhozjan, 2016-01-05:
Thanks Shawn, indeed I will directly customize the runScript.sh

Sorry if Goal 2 wasn't clear, I had well understood the BEGIN and END purpose and I was looking for a way to exclude some periods of time (for instance from 20h to 08h)

Not sure to see how to modifiy a script that never startup at night, except by multiplying the number of lines to describe a whole day as you did on "on_30m_everyday_but_weekend.wpi" file. Is this what you mean?

Thanks,

Benoit

shawnu, 2016-01-05:
Yes Benoit, that is what I mean. Multiplying some lines in script seems not so effective, but you can write your program to generate the entire script. We don't want to introduce some advanced syntax in the schedule script, as we want to keep the parser simple.

closed 2016-02-22 by shawnu