uugear/Witty-Pi #9: Rpi 2 with led red after shutdown

saulgodino, 2016-02-10:
I'm using witty-pi with Rpi 2 running retropie 3.5 (Jessie). The problem is when shutdown the system pressing the button or menu shutdown, after the correct shutdown the led keep red.

shawnu, 2016-02-11:
If the power of RPi was not fully cut, most probably it was because TXD pin gets pulled up.

Please monitor the voltage of TXD pin (GPIO-14 in BCM naming), using multimeter or oscilloscope. This pin should go to 0V after the system has been shutdown.

saulgodino, 2016-02-21:
Using a multimeter I certificate that the TXD pin (GPIO-14) go to 0V after shutdown, but the led of rpi 2 continue red. Only go off after I press the button again. How do I fix it?

shawnu, 2016-02-22:
I am not sure yet, but maybe you could try reducing R5 value to see if it helps. Usually it should be a value in range 6.8K~10K.

If changing resistor is not possible for you, you could send the board back and we will check and fix it. Please contact our support via email first, thank you.

saulgodino, 2016-04-03:
Hello, I can't replace the resistor and I'm from Brazil and is very difficult send the board to you. I tested with a rpi 3 and before install the witty pi software, after the shutdown of raspbian, the power of rpi3 was fully cut, but after install the witty pi software, after shutdown was not fully cut. Maybe it is a software question. Another question is with witty pi appears a rainbow square, but I am using the official power source 2,5A. Without the witty pi does not appears the rainbow square. Sorry for my English.