uugear/Witty-Pi #10: Witty Pi RTC Clock

binhsonvo, 2016-03-10:
I just bought 30 Witty Pi for a communication projects. I install the software from newly set up Rasbian Pi. We have 2 issues:

The hardware clock doesn't work in all board. i2cdetect show 68 and sudo hwclock -r saying "Cannot access the Hardware Clock via any known method"
Brand new board, never install any other software), when turn on the system (after install Witty Pi software) by pressing WittyPi's "on" the system was on in about 1 second, then come back to off, WittyPi breathing led blinks. So fast that Pi just show the rainbow screen, then all off.

Can you share your thought?

bbillp, 2016-03-11:
Remove battery to reset RTC then go into WittyPi folder and delete schedule file, reinstall battery and boot with cat5 to live internet. Now follow instructions to set the RTC time.

binhsonvo, 2016-03-11:
Thanks for the reply, it doesn't work.

Anyway, I sort out some of the issues, and only one remain before I move forward.

RTC doesn't work. If I manually put the command:

sudo modprobe rtc-ds1307

then it works, but there is no way I can make this command in rc.local or modules or profiles. It simply doesn't run this, unless I run it at the Pi prompt.

Any help please?

Thank you

bbillp, 2016-03-11:
I have 3 WittyPis on 3 new Pi 2b in Astrophotography application, I ran the installation per the instructions and everything worked.  I only ran into trouble when I programmed the WittyPi off and on then manually turned the power off and on, the WittyPi does not like that and must be reset by removing the coin battery and erasing the schedule file if any.  I do remember a I2C error but it was my fault caused by not aligning the socket pins when working late one night.    As a passing thought, check the battery voltage.

shawnu, 2016-03-11:
As discussed in emails, this issue is because Raspbian Jessie Lite doesn't have git installed by default, and hence the wiringPi install failed. The problem can be solved by installing git and then run the install script again.

The future version of install script will take care of the missing of git, and avoid this kind of issue.

closed 2016-03-11 by shawnu