uugear/Witty-Pi #11: WittyPi boots RaspberryPi at the right time with an incorrect RTC time

abuchholz, 2016-04-05:
This is a really strange issue and it is somewhat difficult to explain but I will try my best.  I successfully installed a relatively complicated wittyPi schedule:

BEGIN	2015-04-04 00:00:00
END	        2025-07-31 23:59:59
ON H1
OFF H7
ON H1
OFF H1
ON H1
OFF H4
ON H1
OFF H1
ON H1
OFF H1
ON H1
OFF H4

with the intention of repeating daily, on for one hour from midnight to 1AM, off until 8AM, on/off every couple hours with a break in the afternoon.  It works beautifully - except for one very strange thing.  After a day of working correctly the wittyPi shuts down the raspberry pi at 1AM and correctly turns it back on at 8AM.  However, the real time clock is still set to 1AM which is then used to set the raspberry pi clock (incorrectly) and throws the whole schedule off.

To boil down the issue to one simple statement: the witty pi is correctly starting the raspberry pi on time, but it's real time clock is still set to the previous shutdown time. 

Any help is greatly appreciated!  Thanks.

uugear, 2016-04-06:
I have a feeling that it was the "fake-hwclock" package did the bad things.

Can you try running these commands in your RPi's console to see if they help?

```
sudo update-rc.d fake-hwclock remove
sudo apt-get remove fake-hwclock
sudo rm /etc/cron.hourly/fake-hwclock
sudo rm /etc/init.d/fake-hwclock
```

Grimthorr, 2016-04-07:
Conversely, I have recently encountered a similar issue but in reverse: after manually changing the system and RTC time to account for the UK daylight saving time (minus 1 hour), I schedule a 6am startup on my WittyPi but it doesn't turn on until 7am. After boot, the RTC time shows the correct time as 7am. Do you think this could be caused by the `fake-hwclock` too?

uugear, 2016-04-07:
Hi Grimthorr, I don't think that relates to the fake-hwclock, although you should remove it once you get a real RTC connected.

I am aware of the issue you mentioned, it is because the RTC doesn't know about the daylight saving time, while the system date knows about it. As a result the calculation will be incorrect if the daylight saving time boundary appears between the script begin time and current time. The workaround is to move the begin time of the script to some date that already uses the daylight saving time.

Grimthorr, 2016-04-07:
Thanks for the quick reply @uugear!

Does this also apply when setting the startup time manually? I'm not currently using a schedule script but instead I'm calling the following commands from an external script.
```
i2cset -y 0x01 0x68 0x07 $sec
i2cset -y 0x01 0x68 0x08 $min
i2cset -y 0x01 0x68 0x09 $hour
i2cset -y 0x01 0x68 0x0A $date
```
Where `$sec`, `$min`, `$hour` and `$date` are base-16 encoded decimals (so a startup at 6am on 08/04/16 would be `0`, `0`, `6`, `8`) - this worked fine and the WittyPi would startup on time before I adjusted the clocks for daylight saving time.

abuchholz, 2016-04-13:
Thanks for all the help @uugear.  While we think the removal of the fake-hwclock helped, it looks like it might have brought up another issue.  We are working through that now - I will post when we have something written up about it!