uugear/Witty-Pi #12: Reboot instead of shut down

adamkansas, 2016-06-01:
I'm really stumped on this one. I have two separate Pi 3 B+ running with witty Pi boards. One works fine so far. The other, which has the same image on the SD Card, will only reboot regardless of how I attempt to shut it down. The power button, sudo shutdown (-h 0, -h now) all result in the unit rebooting rather than shutting down. If I plug the power directly into the Pi, then it will shut down, but not while the power is run to the witty Pi board. Any thoughts?

uugear, 2016-06-01:
One of our customers met a similar situation. He had the official Raspberry Pi touchscreen connected to RPi (pull power directly from the +5V pin) and kept encountering the reboot during the shutdown. 

He has oscilloscope to monitor the +5V pin and some other signal pins. It turns out that when Witty Pi tried to cut the power, the touchscreen somehow back-feeds the Raspberry Pi (via the +5V pin) and as a result, Raspberry Pi gets enough voltage again to startup and quickly pull up the TXD pin, which cancels the previous process of cutting power. It seems the official touchscreen is a inductive load, which cause this kind of problem. However when I bought an official touchscreen and tried by myself, I couldn't reproduce the problem at all.

Not sure if you are in the same situation, but maybe these information could give you some clues?

adamkansas, 2016-06-03:
Thank you for your help. I was able to get it fixed!

closed 2016-06-03 by adamkansas

uugear, 2016-06-03:
You are welcome. If you are willing to share your solution, please kindly do so since it could help other users that may encounter similar issues.

bbillp, 2016-06-03:
Are these last WittyPi problems specific to the Pi 3B?  Since using the instructions to remove the fake hardware clock I've not noticed problems with the Pi 2B even after a few weeks but I modified my WittyPi by adding a CR123 battery in parallel with the coin battery to extend clock time to months when no power is applied.  I am using a recent version of Jessie with apt-get upgrades and WittyPi 2.14 with the fake hard ware clock removed.  The Pi runs crontab raspistill jobs and my command line login is via SSH so by the time I log in might be 5 minutes or 5 hours and the date and time is correct. I'll keep you posted on the other two model 2Bs with WittyPi RTC.

uugear, 2016-06-03:
I don't think so. This kind of issue is very hard to reproduce. I have never reproduced it on any model of Raspberry Pi.

It is normal that you don't encounter this problem.

> Are these last WittyPi problems specific to the Pi 3B?

uugear, 2016-06-24:
To help people who encounter the same issue, I hereby reference some solutions discovered by our customers.

As described above, the problem is that Raspberry Pi official touchscreen may back-feed power to Raspberry Pi's +5V pin, which causes Raspberry Pi to reboot. So if you connect a diode between the RPi +5V pin and the touchscreen +5V power line, it will avoid the unexpected back-feeding during the shutdown.

Also there is a software solution provided by Alexis Johnson (one of our customers), and I think it is the best solution for this issue, as it doesn't need any hardware modification. He modified the "/etc/init.d/wittypi" file and in line 26 ("stop" section) he added:

`echo "1" > /sys/class/backlight/rpi_backlight/bl_power`

By doing so the touchscreen backlight will turns off immediately during the shutdown process, and will not cause the unexpected power back-feeding.