uugear/Witty-Pi #13: RTC Time changing

adamkansas, 2016-06-03:
I'm having an intermittent issue with the Witty Pi. I have three units running the exact same image, hardware and Witty Pi. From yesterday to today, only two of them had this problem, but it has been occurring for weeks. 

I have the Witty Pi scheduled to turn on at 7:50 CST and turn off at 17:00 CST. It does this fine, but then the next day when the unit starts up, the system time will show 18:00, then after a few moments it will change to the correct time of 7:50, then after another few moments, it changes to 17:00.  When I check the RTC time, it is at the incorrect 17:00.

It really feels like the RTC is not advancing from when it shuts down and then overwrites the system time. I've updated my time zone more times than I can count and in more ways than I care to say. If I run the Witty Pi script in the few moments that the system clock is correct, I can write the time to the RTC and all is well until anywhere from one to several days later, the same issue occurs. I have replaced the battery on the Witty PI several times with no benefit. 

I know I am not explaining this very clearly, but you have been very helpful so far, so I'm hoping you can decipher when I'm saying. Thank you for your help!

uugear, 2016-06-03:
If Witty Pi wakes up your RPi at correct time, it means the RTC is keeping the time properly, otherwise it would not boot RPi at the right moment. What you described seems to be a time synchronization problem, and somehow the RTC time was override by incorrect system time (which I have no idea where is it from yet). You should be able to figure out what happened by checking the wittyPi.log file and the system log file.

If possible, it is better to use the latest version of Witty Pi software (V2.18 by now). The syncTime.sh has been updated and can synchronize the time more smartly. It will even take the NTP time if Internet connection is available.

uugear, 2016-06-03:
After a second thought, I think you can try these two steps:

1. Update software to V2.18
2. Remove the "fake-hwclock" package

laser74, 2016-06-06:
Hi, i am also also having some issues with 2 witty Pis.  Both running the latest Jessie on RPi 2 and fully updated.  My issue was the RTC time was dropping back to the last shutdown time.  I have unstalled the fake-hwclock package, and updated to V2.18 of the witty pi software.  

Now I am observing that if the pi is not connected to the internet, the system time is not updated.  

whittypi.sh displayed the following;

>>> Your system time is: Thu 01 Jan 1970 08:01:17 AWST
hwclock:  The Hardware Clock registers contain values that are either invalid (e.g. 50th day of month) or beyond the range we can handle (e.g. Year 2095)
>>> Your RTC time is:

wittyPi.log contains;

[1970-01-01 08:00:10] Witty Pi daemon (v2.181) is started.
[1970-01-01 08:00:11] File "schedule.wpi" not found, skip running schedule scrip
t.
[1970-01-01 08:00:21] Pending for incoming shutdown command...
[1970-01-01 08:01:10] Synchronizing time between system and Witty Pi...
[1970-01-01 08:01:12] RTC contains newer time.
[1970-01-01 08:01:12] Writing RTC time to system...
[1970-01-01 08:01:13] Failed :-(
[1970-01-01 08:01:13] hwclock: The Hardware Clock does not contain a valid time,
 so we cannot set the System Time from it. Unable to set system clock.

If I plug the pi into my router and restart, the time is updated after a minute or so.

wittyPi.log contains;

[1970-01-01 08:00:10] Witty Pi daemon (v2.181) is started.
[1970-01-01 08:00:11] File "schedule.wpi" not found, skip running schedule scrip
t.
[1970-01-01 08:00:21] Pending for incoming shutdown command...
[2016-06-06 16:20:32] Synchronizing time between system and Witty Pi...
[2016-06-06 16:20:33] Writing RTC time to system...
[2016-06-06 16:20:33] Failed :-(
[2016-06-06 16:20:33] hwclock: The Hardware Clock does not contain a valid time,
 so we cannot set the System Time from it. Unable to set system clock.
[2016-06-06 16:20:34] Internet detected, apply NTP time to system and Witty Pi..
.
[2016-06-06 16:20:34] Pushing NTP to update system time...
[2016-06-06 16:20:41] Writing system time to RTC...

wittypi.sh now shows that the RTC clock has been updated;

Now if I restart without the connection to the router, the system time is not updated (still Jan 1, 1970)

This is happening on 2 new witty I have checked the battery voltages and they are 2.97 Volts.

Thanks...

uugear, 2016-06-06:
Hi laser74, after removing the fake-hwclock, the system time will go to year 1970 every time you turn on your Raspberry Pi, and the RTC should write its time into the system, the process should look like these:

> [1970-01-01 01:00:30] Witty Pi daemon (v2.181) is started.
[1970-01-01 01:00:34] Schedule next shutdown at: 2016-06-06 11:35:00
[1970-01-01 01:00:35] Schedule next startup at: 2016-06-06 11:36:00
[1970-01-01 01:00:37] Synchronizing time between system and Witty Pi...
[1970-01-01 01:00:39] RTC contains newer time.
[1970-01-01 01:00:39] Writing RTC time to system...
[2016-06-06 11:33:29] Done :-)
[2016-06-06 11:33:30] Pending for incoming shutdown command...

In your log I can see the complain about the wrong value in RTC. Not sure how those wrong values got into the RTC but in the major of case, you can fix them by writing the correct system time into your RTC.

So if you boot up your RPi with Internet connection, and the system time is correct, then you can choose the "1. Write system time to RTC" option after running wittyPi.sh. Then you can remove your Internet connection and reboot, and during the next booting you should get similar logs as I shown above, and both your system time and RTC time should be correct.

laser74, 2016-06-06:
Hi,  thanks for you swift reply, I had made some further progress...

The main problem was that even though I was updating the RTC time as you described, on every restart the RTC clock was corrupted.  I found that the culprit is the ntpd daemon, which has been reported by others as corrupting the saved time on various RTC clocks.  I have disabled this daemon at startup via;

sudo update-rc.d -f ntp remove

The syncTime.sh script still works for updating via ntp when the network is connected.  

The time appears to be stable now, so have started a script running tonight...

uugear, 2016-06-06:
Thanks for the information laser74. We have received multiple reports about RTC date corruption but we could not reproduce it so far. I guess not every version of ntpd daemon has this problem. Do you have a link that contains more information about this ntpd daemon issue? Thanks in advance.

adamkansas, 2016-06-06:
I believe that I have fixed the issue in my particular application. It seems as though sometimes the system time had not yet been updated when the synctime.sh script was executed. I have delayed it from 60s after boot to 240s after boot to give the system plenty of time to update its time. This has kept my issue at bay. However if the internet connection isn't available when the unit starts, it will create issues.

adamkansas, 2016-06-06:
Also, I just attempted to update to 2.18, but after doing so, it still displays as 2.16. What is the proper way to update?

uugear, 2016-06-06:
Delay running the syncTime.sh is a workaround, it gives enough time for NTP to update your system. However as you mentioned, it only works when Internet connection is there. Removing fake-hwclock is more thorough solution, but some users reported the date on RTC will be corrupt after reboot (which I could not reproduce yet). laser74 pointed that data corruption is caused by ntpd daemon, which is new to me and really want to learn more details.

If you want to use new version, the best way is to rename the wittyPi directory, and then run the commands below:

`wget http://www.uugear.com/repo/WittyPi/installWittyPi.sh`
`sudo sh installWittyPi.sh`

bbillp, 2016-06-10:
Is it possible these problems might be related to a bad circuit from the
Witty Pi coin battery to the RTC chip ?  I have one Witty Pi of Three all
on Pi-2B that fails often but not every boot even when staying connected to
the CAT-5 cable and with USB power always on, just the white LED blinking.
After boot and waiting extra minutes DATE give 1969.  I would like to
monitor the voltage on the RTC.  Is it possible to get this part of the
circuit diagram to read ?

uugear, 2016-06-10:
If the RTC loses power or something, it should go back to year 2000 (or 1999 sometimes) instead of 1969. The year 1969 is a corrupt data and could be caused by the ntpd daemon, according to laser74's comment. Maybe you can try disabling the ntpd daemon at boot:

> sudo update-rc.d -f ntp remove

If you want to check the power supply of the RTC chip, you can check this diagram:
[image]

uugear, 2016-06-10:
BTW, one of our customers said that removing the fake-hwclock and then reinstall it can also workaround this kind of issue. I could not confirm that since I can't reproduce the issue yet. But I guess it is another direction to try, if other approaches can not work out.

bbillp, 2016-06-10:
Thank you for the diagram, tracing should be easy.

RTC data is corrupt re the message in WittyPi.sh

All my Pi-2B has the most recent Jessie load and WittyPi updates this week with the fake hw-clock removed.

Two of three Witty Pis seem to work fine, all three Pis normally work from a free standing router, no internet, the router is used in the hills for Pi Astrophotography but the test mentioned here are in my home using the main internet router.

I will remove the NTPD daemon on this Pi and try again, the condition occurs 1 out of 3 shutdown/reboot cycles. No WittyPi schedules during these tests.

Have a good weekend ... Bill

bbillp, 2016-06-10:
Obviously a very difficult problem, well maybe another product could be a GPS interface for time.  

I have a suggestion for the WittyPi, a pin for external shutdown for example:  the batteries on my hill top PiCamera go low and the battery interface sends a signal to the WittyPi to power off .   I suppose some people would want a signal to force the WittyPi to power On, I think this external pin would be a great addition to the features.       Bill.

bbillp, 2016-06-10:
Good news, add 'remove ntp' procedure to the remove hwclock procedure, this is the first time I could boot and keep the correct time on THIS raspberry Pi 2B.  I have booted at least 10 times with and without the internet and with and without USB power on, the last 2 boots without Internet and without USB power so this is very good.  I'll report Monday after long term power off .

The BAT diode has a .2 volt drop so the battery shows 3.0 volts and the RTC pin 8 shows 2.75volts. What is the lowest voltage at pin 8 to maintain the time ?  I can swap the 2032 battery for a A123 or 2 AA Alkaline batteries at the same voltage for months of backup.   
Bill

uugear, 2016-06-10:
Hi Bill, we will break out the switch signal line in future version. At the time being, you can solder wires on the button to break them out. 

The push button on Witty Pi has two pins, and one of them connects to GND. You should be able to use a N-Channel MOSFET to work as a electronic switch. After connecting the source of MOSFET to ground, and drain of MOSFET to the other pin of the button, giving a positive pulse to the gate of MOSFET should trigger your Witty Pi (equals to pushing the button).

According the datasheet of DS1337, even the voltage drop to 1.3V, it can still make the time keeping. So I think your battery is good enough.

azltx, 2016-10-16:
Thanks this thread, the hwclock finally not complain about having a invalid time... after apply the suggestions above:
- sudo apt remove fake-hwclock
- sudo update-rc.d -f ntp remove