uugear/Witty-Pi #14: witty pi not rebooting or turning on

Jennyuni, 2016-06-13:
Hi, I am using a raspberry pi 3 and the witty pi code 2.181.

The pi won't turn on when pressing the on button and won't reboot after a scheduled shut down. It will shut down when scheduled and shut down when the button is pressed. Could you advise me what the problem could be?

Thanks

uugear, 2016-06-13:
Hi Jennyuni,

It is better to provide the wittyPi.log file content here to rule out some possible problems.

Before your Raspberry Pi shutdown, did you see the white LED on Witty Pi lighted up? If not, it means the Witty Pi daemon is not running properly. It should be automatically run as root user. In this case you should check the content of /etc/init.d/wittypi file.

I may be more helpful if you could provide the log file.

Jennyuni, 2016-06-14:
Hi Shawn,

Thank you so much for the quick response.

Yes the white LED light came on when it shut down.  I have provided the witty Pi log file and the /etc/init.d/wittypi file.

Many thanks,

Jenny

uugear, 2016-06-14:
Sorry but I don't see your log file. If you are replying with email, you may need to paste its content into your email body.

Jennyuni, 2016-06-14:
Hi, I attached the file but here it is pasted out:

[2016-06-13 13:29:18] Witty Pi daemon (v2.181) is started.
[2016-06-13 13:29:18] Witty Pi is not connected, skipping I2C communications...
[2016-06-13 13:29:19] File "schedule.wpi" not found, skip running schedule script.
[2016-06-13 13:29:24] Witty Pi is not connected, skip synchronizing time...
[2016-06-13 13:29:28] Pending for incoming shutdown command...
[2016-06-13 13:31:33] Witty Pi daemon (v2.181) is started.
[2016-06-13 13:31:35] File "schedule.wpi" not found, skip running schedule script.
[2016-06-13 13:31:39] Synchronizing time between system and Witty Pi...
[2016-06-13 13:31:41] RTC time has not been set before (stays in year 1999/2000).
[2016-06-13 13:31:41] Writing system time to RTC...
[2016-06-13 13:31:43] Done :-)
[2016-06-13 13:31:43] Pending for incoming shutdown command...
[2016-06-13 13:37:13] Writing system time to RTC...
[2016-06-13 13:37:14] Done :-)
[2016-06-13 13:39:21] Seting startup time to "?? 14:00:00"
[2016-06-13 13:39:21] Done :-)
[2016-06-13 13:40:13] Seting shutdown time to "?? 14:10"
[2016-06-13 13:40:14] Done :-)
[2016-06-13 13:42:25] Witty Pi daemon (v2.181) is started.
[2016-06-13 13:42:27] File "schedule.wpi" not found, skip running schedule script.
[2016-06-13 13:42:31] Synchronizing time between system and Witty Pi...
[2016-06-13 13:42:33] RTC contains newer time.
[2016-06-13 13:42:33] Writing RTC time to system...
[2016-06-13 14:09:29] Done :-)
[2016-06-13 14:09:30] Pending for incoming shutdown command...
[2016-06-13 14:10:00] Shutdown command is received...
[2016-06-13 14:10:00] Halting all processes and then shutdown Raspberry Pi...
[2016-06-13 14:10:03] Witty Pi daemon (v2.181) is started.
[2016-06-13 14:10:05] File "schedule.wpi" not found, skip running schedule script.
[2016-06-13 14:10:09] Synchronizing time between system and Witty Pi...
[2016-06-13 14:10:11] RTC contains newer time.
[2016-06-13 14:10:11] Writing RTC time to system...
[2016-06-13 14:23:14] Done :-)
[2016-06-13 14:23:15] Pending for incoming shutdown command...
[2016-06-13 14:29:26] Clearing auto startup time...
[2016-06-13 14:29:26] done :-)
[2016-06-13 14:29:33] Clearing auto shutdown time...
[2016-06-13 14:29:33] done :-)
[2016-06-13 14:29:48] Writing RTC time to system...
[2016-06-13 14:29:49] Done :-)
[2016-06-13 14:45:07] Seems Witty Pi board is not connected? Quitting...
[2016-06-13 14:45:27] Seems Witty Pi board is not connected? Quitting...
[2016-06-13 14:45:43] Shutdown command is received...
[2016-06-13 14:45:43] Halting all processes and then shutdown Raspberry Pi...
[2016-06-13 14:45:48] Witty Pi daemon (v2.181) is started.
[2016-06-13 14:45:49] File "schedule.wpi" not found, skip running schedule script.
[2016-06-13 14:45:54] Synchronizing time between system and Witty Pi...
[2016-06-13 14:45:55] Writing RTC time to system...
[2016-06-13 14:48:01] Done :-)
[2016-06-13 14:48:01] Internet detected, apply NTP time to system and Witty Pi...
[2016-06-13 14:48:01] Pushing NTP to update system time...
[2016-06-13 14:48:03] Pending for incoming shutdown command...
[2016-06-13 14:48:06] Writing system time to RTC...
[2016-06-13 14:48:08] Done :-)
[2016-06-13 14:52:28] Seting shutdown time to "?? 15:01"
[2016-06-13 14:52:28] Done :-)
[2016-06-13 14:53:02] Seting startup time to "?? 15:06:01"
[2016-06-13 14:53:02] Done :-)
[2016-06-13 15:01:00] Shutdown command is received...
[2016-06-13 15:01:00] Halting all processes and then shutdown Raspberry Pi...
[2016-06-13 15:01:05] Witty Pi daemon (v2.181) is started.
[2016-06-13 15:01:06] File "schedule.wpi" not found, skip running schedule script.
[2016-06-13 15:01:11] Synchronizing time between system and Witty Pi...
[2016-06-13 15:01:12] Internet detected, apply NTP time to system and Witty Pi...
[2016-06-13 15:01:12] Pushing NTP to update system time...
[2016-06-13 15:01:15] Pending for incoming shutdown command...
[2016-06-13 15:17:35] Writing system time to RTC...
[2016-06-13 15:17:36] Done :-)
[2016-06-13 15:21:47] Clearing auto startup time...
[2016-06-13 15:21:47] done :-)
[2016-06-13 15:21:57] Clearing auto shutdown time...
[2016-06-13 15:21:57] done :-)
[2016-06-13 15:22:22] Seting shutdown time to "?? 15:30"
[2016-06-13 15:22:22] Done :-)
[2016-06-13 15:23:00] Seting startup time to "?? 15:40:00"
[2016-06-13 15:23:00] Done :-)
[2016-06-13 15:30:00] Shutdown command is received...
[2016-06-13 15:30:00] Halting all processes and then shutdown Raspberry Pi...
[2016-06-13 15:30:05] Witty Pi daemon (v2.181) is started.
[2016-06-13 15:30:06] File "schedule.wpi" not found, skip running schedule script.
[2016-06-13 15:30:10] Synchronizing time between system and Witty Pi...
[2016-06-13 15:30:11] Writing RTC time to system...
[2016-06-14 09:18:14] Done :-)
[2016-06-14 09:18:14] Internet detected, apply NTP time to system and Witty Pi...
[2016-06-14 09:18:14] Pushing NTP to update system time...
[2016-06-14 09:18:16] Pending for incoming shutdown command...
[2016-06-14 09:18:14] Writing system time to RTC...
[2016-06-14 09:18:15] Done :-)
[2016-06-14 09:23:37] Seting shutdown time to "?? 09:26"
[2016-06-14 09:23:37] Done :-)
[2016-06-14 09:24:30] Seting startup time to "?? 09:29:00"
[2016-06-14 09:24:31] Done :-)
[2016-06-14 09:26:00] Shutdown command is received...
[2016-06-14 09:26:00] Halting all processes and then shutdown Raspberry Pi...
[2016-06-14 09:26:04] Witty Pi daemon (v2.181) is started.
[2016-06-14 09:26:06] File "schedule.wpi" not found, skip running schedule script.
[2016-06-14 09:26:10] Synchronizing time between system and Witty Pi...
[2016-06-14 09:26:12] RTC contains newer time.
[2016-06-14 09:26:12] Writing RTC time to system...
[2016-06-14 09:33:34] Done :-)
[2016-06-14 09:33:35] Pending for incoming shutdown command...

uugear, 2016-06-14:
Thanks for the log.

What if you only schedule the startup and don't schedule the shutdown, then you press the button to shutdown and wait? Will your Witty Pi wake up later?

Jennyuni, 2016-06-14:
Hi,

No the Pi doesn’t turn on again. I scheduled a start up then shut down the pi using the button. The pi didn’t come back on.

uugear, 2016-06-14:
Do you notice any physical damage around the RTC chip (U3)? 

I am afraid it is a hardware fault. Please kindly send an email to support@uugear.com, to arrange a fixing/replacement.

uugear, 2016-07-01:
After sending back the Witty Pi unit and we checked on it, we confirm the unit is not faulty. It passes all auto testing cases and manual test cases, so its functionality is complete and correct. We send the unit back to customer.

There must be something else that cause the issue. If this issue can still be reproduced and you find out the actual reason that causes it, please share it here, thank you.

closed 2016-09-07 by uugear

thorntj91, 2016-09-13:
I have exactly the same issue on a fresh install of Raspbian. My log looks identical to Jennyuni's. 

My wittyPi will shutdown the Pi via the button and a scheduled shutdown event. 
The white LED is on when it shuts down. 
After this it does not respond to scheduled start ups or activation manually via the button.

dphandy, 2016-12-13:
Was this issue resolved?  I, too, can schedule a shutdown OK, but the start up time does not work. The white LED is on at shutdown.

thorntj91, 2016-12-20:
Unfortunately not. I was hope that uugear would provide from input regarding this but no luck. I will make an effort to reach out to them and reference this thread. It seems clear that this in a tangible issue.  It may work on their test bench, but it's no good if it doesn't work in the field. This didn't work with a Pi that was fresh out of the box. 
If you have any success do post back here. I'll do the same :)

Jennyuni, 2016-12-20:
Mine was to do with the power supply I think. I can't remember exactly and I find that sometimes the time on the witty pi is different to the raspberry pi. Because I was using a battery pack I needed a constant small supply of power going to the pi in order to turn it on. This was resolved by using a dummy resistor.

calgradstudent, 2017-06-15:
I am having this same issue, but I am not using a battery pack in my case. Has anyone else resolved this problem? I am running Raspbian Jessie Lite and using a Raspberry Pi A+.

merimem, 2017-06-29:
Having the same problem too!! Any idea how to resolve this?

uugear, 2017-06-29:
We noticed that some customers connect the power supply directly to Raspberry Pi, and have Witty Pi mounted on Raspberry Pi at the same time. This configuration is incorrect, and as a result, Witty Pi's software can shutdown the system, but Witty Pi can not turn on Raspberry Pi, since the power is not controlled by Witty Pi.

The correct configuration is to connect power supply to Witty Pi only, and let Witty Pi to power your Raspberry Pi.

galatgithub, 2017-08-22:
I have the same problem. The RPi does not start at the scheduled time by the WittyPi.
My power plug is well connected to the Witty Pi board.