uugear/Witty-Pi #17: Update to RTC time

Yeocti, 2016-08-24:
In wittyPi.sh it seems that time taken as argument to set_auto_startup and set_auto_shutdown functions is converted to UTC time. However in utilities.sh and system_to_rtc function, RTC time is not automatically updated to UTC time.

Ex :
- If system is set to local time, RTC time will be updated to local time. As a consequence, system will not start/shutdown at schedule time.
- If system is set to UTC time, everything is working fine.

uugear, 2016-08-27:
Hi Yeocti, have you seen the consequence when setting local time to non-UTC? My system is set to CET time, but I never saw the consequence you mentioned, and I can schedule startup/shutdown normally.

Yeocti, 2016-08-27:
Hi,

My system time is set to local time (Europe/Paris). When I launch wittyPi.sh and use option 1 (system time to RTC) RTC is set to local time. However, when I schedule shutdown or startup, the time is converted to UTC time and as a result shutdown or startup does not happen at specified time. The only way to get it working normally on my system was to make sure RTC was set to UTC time.

Looking at the code it seems to me there is a problem because :
- In wittyPi.sh, functions set_auto_startup and set_auto_shutdown make use of the function get_utc_date_time so the RTC alarms are set in UTC time.
- In utilities.sh, system_to_rtc does not set RTC time to UTC resulting in a possible discrepancy between time zone used for RTC time and time zone used for RTC alarms.

Is this happening only on my system ?
On yours, your local time is to CET. What happen when you use option 1 (system time to RTC), is your RTC clock set to CET or to UTC ?

uugear, 2016-08-28:
My RTC is set to CET after running option 1, and the startup/shutdown alarm still works as expected.

I think your analyzing makes sense, but in reality we didn't see the problem you mentioned. Also none of our customers complain this before. I think at least in Raspbian (Wheezy and Jessie) we could not reproduce this issue.

Are you seeing this issue in Alpine Linux?

Yeocti, 2016-08-28:
This is indeed Alpine Linux related and I found the issue.

Basically, hwclock is provided by busybox. In that case, hwclock does not make any conversion to time. `hwclock -w` directly write system time to RTC. If system is set to local time, RTC will also be set to local time. If you want RTC to be set to UTC time, you have to use `hwclock -wu`.

I installed util-linux which provide the 'full' hwclock utility. Using this one, the issue does not occur.
It seems that the necessary conversion are made in background according to the configuration file. As a result `hwclock -w` will set RTC to UTC time if `clock="UTC"` in configuration file (which is the case in default configuration file). I also tried with `clock=local` in configuration file. The issue did not occured.

So, I think this issue is closed :)

closed 2016-08-28 by Yeocti