uugear/Witty-Pi #19: Creating i2c device gives an error

Yeocti, 2016-08-24:
I was trying to use wittyPi on Alpine Linux.

Issuing more than once the command `echo ds1307 0x68 > /sys/class/i2c-adapter/i2c-1/new_device` gives the following error (in fact the device already exists when this error occurs) `sh: write error: Invalid argument`

Currently, I modified utilities.sh so that the load_rtc function check the existence of /dev/rtc0 but I don't know if it's correct.

```
load_rtc()
{
  modprobe rtc-ds1307
  if [ ! -c /dev/rtc0 ] ; then
    local output=$((sh -c 'echo ds1307 0x68 > /sys/class/i2c-adapter/i2c-1/new_device') 2>&1)
    if [ ! -z "$output" ] && [ "sh: echo: I/O error" != "$output" ] ; then
      log "$output"
    fi
  fi
}
```

uugear, 2016-08-27:
I guess it only happens on Alpine Linux, and it might relates to the shell implementation.

In Raspbian this function works well, and if the device already exists, the error message "sh: echo: I/O error" will be ignored.

Yeocti, 2016-08-27:
Fine. This is what I thought.
I will keep my modified function while running Alpine Linux.

Thanks.

bbillp, 2016-08-27:
Where does 'last log in time' come from ?   When I use the Pi2+WittyPi+ last updates WITHOUT a router date command displays the correct time and date but when I come home to download and attach the CAT5 router cable with Internet and log in the 'last time' is Dec 31, 1969.  Sudo ./WittyPi.sh shows the correct time/date for today.  Date command is correct so where does the 'last log in time' come from? 
It's not a problem, just curious .... Bill