uugear/Witty-Pi #20: WittyPi does not shutdown Rpi

sebgosse, 2016-09-06:
Hello,
I have a WittyPi 2, a Rpi zero (with a Zero4u), a 3G key connected.
I have a Script that keep on Rpi during 5min, keep off Rpi during 1min?
However, the Rpi does not shutdown everytime.
For example, it should shutdown at 00:23 but never shutdown. I need to shutdown manually, and I wake up it manually at 00:30.
I have the NOOBS_V1_9 configuration.

Logs :

[1970-01-01 01:00:17] Witty Pi 2 daemon (v2.52) is started.
[1970-01-01 01:00:20] Synchronizing time between system and Witty Pi...
[1970-01-01 01:00:21] Writing RTC time to system...
[2016-09-07 00:18:36] Done :-)
[2016-09-07 00:18:40] Schedule next shutdown at: 2016-09-07 00:23:00
[2016-09-07 00:18:42] Schedule next startup at: 2016-09-07 00:24:00
[2016-09-07 00:18:43] Pending for incoming shutdown command...
[1970-01-01 01:00:20] Witty Pi 2 daemon (v2.52) is started.
[1970-01-01 01:00:25] Synchronizing time between system and Witty Pi...
[1970-01-01 01:00:25] Writing RTC time to system...
[2016-09-07 00:30:18] Done :-)
[2016-09-07 00:30:22] Schedule next shutdown at: 2016-09-07 00:35:00
[2016-09-07 00:30:24] Schedule next startup at: 2016-09-07 00:36:00
[2016-09-07 00:30:25] Pending for incoming shutdown command...
[2016-09-07 00:30:28] Internet detected, apply NTP time to system and Witty Pi...
[2016-09-07 00:30:37] Writing system time to RTC...
[2016-09-07 00:30:39] Done :-)

uugear, 2016-09-07:
Did you mean the auto shutdown never work? Or sometimes it works, sometimes not?

The log shows there was no shutdown command received. If you have a voltage meter, maybe you can verify if the shutdown alarm has occurred or not.

In the 6-pin header, there is an INT pin(the 3rd pin from left to right). By default it is in HIGH state, and it will go to LOW if the shutdown/startup alarm occurred. When the scheduled shutdown time comes, and your Witty Pi doesn't shutdown, you can measure the voltage between the INT and GND pin (the last pin in the 6-pin header).

sebgosse, 2016-09-07:
Thank you for your quick reply @uugear !
Actually, It works sometimes, sometimes not.
I continue to investigate with the /var/log/syslog.
I saw that sometimes I catch the shutdown command, but nothing appears after that (I need to shutdown and wake up physically to restart the Pi).
I will measure the INT pin to check the state.

sebgosse, 2016-09-07:
I think that I have found the reason.
My programm send a picture every 6 minutes to a  FTP Server.
With a Wifi key  (connected to Internet), it seems to work perfectly ! 
My problem appear only when my 3G key is used (I use wvdial to connect to Internet).

Is It possible that the wittypi tried to shutdown properly but a process block the shutdown ?

uugear, 2016-09-07:
If the GPIO-4 pin is strongly pulled up (to 3.3V) by hardware, Witty Pi will not be able to shutdown. However I don't think a 3G dongle can do this...

sebgosse, 2016-09-07:
So,
I just checked the INT PIN state when shutdown should occured...
It goes from 5V to 0V correctly, but raspberry keeps ON...
I don't understand :( 
If the INT PIN is 0V, WittyPi should shutdown (without software), right ?
Is there a possible link with the Default Off/On Jumper ?

Thank you for your help :)

uugear, 2016-09-07:
There is a capacitor between the INT and GPIO-4, so pulling INT to 0V will only generate a pulse, which seems not gets caught by the daemon.

I notice your Pi is in OFF state for only one minute, which may be too short. Have you tried changing the OFF state to something like 5 minutes?

sebgosse, 2016-09-09:
Hi,
I tried to test with 5min ON 5min OFF but it is the same. I also tried 30min ON / 30min OFF.
I notice that I see the shutdown command in traces but after the command is received : the White led on the zero4u turns off but the blue led on the card stay on. In this state, impossible to connect via ssh : the USB seems to be unmount. 
Is there a software process that could block the shutdown process ?

uugear, 2016-09-09:
After receiving the shutdown command, the daemon of Witty Pi will run "sudo shutdown -h now" command, which will try to halt all processes and then shutdown Raspberry Pi. Maybe your 3G dongle has launched a process that refused to be halted, and hence the shutdown can not be finished.

Another possibility is that your 3G dongle's driver constantly pulls up TXD pin, which prevents Witty Pi to fully cut the power.

You can verify this easily. With the 3G dongle connected, you just run "sudo shutdown -h now" command and see if it can shutdown your Raspberry Pi properly.

You can also monitor the voltage on TXD pin (using voltage meter or oscilloscope), the TXD pin should go LOW after the shutdown happens, if it doesn't, Witty Pi can not cut the power.