uugear/Witty-Pi #23: Shutdown at start with activate 1 Wire option

OscarCid, 2016-11-24:
i have a DS18B20 sensor, and i need activate the 1-wire option, but when i activate this, the raspberry pi instantly shut down. i can't use the 1-wire option with wittypi?

When I deactivate the option, it works normally.

uugear, 2016-11-25:
Which GPIO pin was connected to your 1-wire sensor? It couldn't be GPIO-4 (BCM naming), since it is used by Witty Pi.

You may change the GPIO pin used by 1-wire bus, or you can change the pin used by Witty Pi: http://www.uugear.com/portfolio/change-the-pin-that-used-by-witty-pi/

OscarCid, 2016-11-27:
Thanks for the reply and sorry for the delay.

The solution was simpler than I thought, by default the raspberry assigns the GPIO 4 (Pin # 7) to 1-wire and that causes the problem, the solution is simply to change the default GPIO used by the 1-wire communication editing The file /boot/config.txt file and add this line:

```
dtoverlay = w1-gpio, gpiopin = x
```
Where x is the GPIO pin you want to use. Restart and this is solved.

moccand, 2017-01-15:
Oscar, which GPIO pin would you advice ?