uugear/Witty-Pi #24: Raspberry is not booting or shutting down

mbaertl, 2016-12-10:
Hello,
I have now tried the witty-pi2 on my raspberry (installed two times) and the wittypi is not doing anything. I can schedule the startup and shutdown, but nothing is happening.

[2016-12-10 08:19:51] Copying "on_5m_every_20m.wpi" to "schedule.wpi"...
[2016-12-10 08:19:51] Running the script...
[2016-12-10 08:19:53] Schedule next shutdown at: 2016-12-10 08:25:00
[2016-12-10 08:19:53] Schedule next startup at: 2016-12-10 08:40:00
[2016-12-10 08:19:54] Done :-)

Nothing happens at 08:25:00 or later. The only errors I see are in the messages file and I have not found any reason for this. I have found some hints about this messages, but nothing was helping or had a relation to my problem:

[60426.913176] i2c i2c-1: Failed to register i2c client ds1307 at 0x68 (-16)
[60435.670685] rtc-ds1307 1-0068: rtc core: registered ds1307 as rtc0
[60435.670752] rtc-ds1307 1-0068: 56 bytes nvram
[60435.716984] i2c i2c-1: Failed to register i2c client ds1307 at 0x68 (-16)
[60438.568459] rtc-ds1307 1-0068: rtc core: registered ds1307 as rtc0
[60438.568549] rtc-ds1307 1-0068: 56 bytes nvram
[60438.604341] i2c i2c-1: Failed to register i2c client ds1307 at 0x68 (-16)

Any idea what is wrong? Could it be related to the HW? 
regards,
Matthias

uugear, 2016-12-13:
Hi Matthias,

I have saw those error messages on units that work properly, so I believe they are not pointing to the real problem.

You haven't pasted the full log, so I am not sure if the daemon.sh is running properly. In the log you should see something like "Pending for incoming shutdown command...", if you haven't found this in the log, it means the daemon is not running properly and Witty Pi can not shutdown your RPi.

mbaertl, 2016-12-13:
Hi,
thanks for your feedback. It really helped me a lot to know that there could be an issue with the deamon. As the deamon should be installed during the setup I was assuming something was going wrong during the install script execution. I have now downloaded the latest install package from the internet and installed it again. I am not sure if I have used the shell "sh" when executing the "installWittyPi.sh" in the past. Now I can see the entries from the deamon in the logs. The shutdown was now working as well. So thanks again. 
br,
Matthias