uugear/Witty-Pi #25: scheduling pi wake up

gmh04, 2017-01-04:
I'm having some trouble with scheduling my pi with witty-pi.  Typically, it runs for a day or two then begins eratrically scheduling next shutdown.  My use case is for the pi to wake up for 2 minutes on the hour between the hours of 9 and 5.  Please find below the context of my .wpi file (perhaps there is a better way of doing this?) and a typical log.

The pi is attached to a battery (which is currently connected to a power source) and has a wifi connection on start up.

```
BEGIN	2016-11-22 09:00:00
END	2025-07-31 23:59:59

# 9
ON	M2	# keep ON state for 2 minutes
OFF	M58	# keep OFF state for 58 minutes

# 10
ON	M2	# keep ON state for 2 minutes
OFF	M58	# keep OFF state for 58 minutes

# 11
ON	M2	# keep ON state for 2 minutes
OFF	M58	# keep OFF state for 58 minutes

# 12
ON	M2	# keep ON state for 2 minutes
OFF	M58	# keep OFF state for 58 minutes

# 13
ON	M2	# keep ON state for 2 minutes
OFF	M58	# keep OFF state for 58 minutes

# 14
ON	M2	# keep ON state for 2 minutes
OFF	M58	# keep OFF state for 58 minutes

# 15
ON	M2	# keep ON state for 2 minutes
OFF	M58	# keep OFF state for 58 minutes

# 16
ON	M2	# keep ON state for 2 minutes
OFF	M58	# keep OFF state for 58 minutes

# 17
ON	M2	# keep ON state for 2 minutes
OFF H15	M58	# keep OFF until 9am
```
```
[2016-12-19 16:58:17] Copying "timelapse.wpi" to "schedule.wpi"...
[2016-12-19 16:58:17] Running the script...
[2016-12-19 16:58:20] Schedule next shutdown at: 2016-12-19 17:02:00
[2016-12-19 16:58:20] Schedule next startup at: 2016-12-20 09:00:00
[2016-12-19 16:58:20] Done :-)
[2016-12-19 17:02:00] Shutdown command is received...
[2016-12-19 17:02:00] Halting all processes and then shutdown Raspberry Pi...
[2016-12-19 17:02:05] Witty Pi daemon (v2.183) is started.
[2016-12-19 17:02:07] Schedule next shutdown at: 2016-12-20 09:02:00
[2016-12-19 17:02:08] Schedule next startup at: 2016-12-20 10:00:00
[2016-12-19 17:02:11] Synchronizing time between system and Witty Pi...
[2016-12-19 17:02:12] RTC contains newer time.
[2016-12-19 17:02:12] Writing RTC time to system...
[2016-12-20 09:00:32] Done :-)
[2016-12-20 09:00:34] Pending for incoming shutdown command...
[2016-12-20 09:02:00] Shutdown command is received...
[2016-12-20 09:02:00] Halting all processes and then shutdown Raspberry Pi...
[2016-12-20 09:02:07] Witty Pi daemon (v2.183) is started.
[2016-12-20 09:02:09] Schedule next shutdown at: 2016-12-20 10:02:00
[2016-12-20 09:02:09] Schedule next startup at: 2016-12-20 11:00:00
[2016-12-20 09:02:13] Synchronizing time between system and Witty Pi...
[2016-12-20 09:02:14] RTC contains newer time.
[2016-12-20 09:02:14] Writing RTC time to system...
[2016-12-20 10:00:32] Done :-)
[2016-12-20 10:00:34] Pending for incoming shutdown command...
[... 159 lines cut ...]
[2016-12-21 17:00:34] Pending for incoming shutdown command...
[2016-12-21 18:00:19] Shutdown command is received...
[2016-12-21 18:00:19] Halting all processes and then shutdown Raspberry Pi...
[2016-12-21 18:00:24] Witty Pi daemon (v2.183) is started.
[2016-12-21 18:00:26] Schedule next shutdown at: 2016-12-22 09:02:00
[2016-12-21 18:00:26] Schedule next startup at: 2016-12-22 10:00:00
[2016-12-21 18:00:30] Synchronizing time between system and Witty Pi...
[2016-12-21 18:00:31] Writing RTC time to system...
[2016-12-22 09:00:32] Done :-)
[2016-12-22 09:00:32] Internet detected, apply NTP time to system and Witty Pi...
[2016-12-22 09:00:32] Pushing NTP to update system time...
[2016-12-22 09:00:34] Pending for incoming shutdown command...
[2016-12-22 09:59:04] Writing system time to RTC...
[2016-12-22 09:59:06] Done :-)
[2016-12-22 10:00:00] Shutdown command is received...
[2016-12-22 10:00:00] Halting all processes and then shutdown Raspberry Pi...
[2016-12-22 10:00:05] Witty Pi daemon (v2.183) is started.
[2016-12-22 10:00:08] Schedule next shutdown at: 2016-12-23 14:02:00
[2016-12-22 10:00:08] Schedule next startup at: 2016-12-23 15:00:00
[2016-12-22 10:00:11] Synchronizing time between system and Witty Pi...
[2016-12-22 10:00:12] RTC contains newer time.
[2016-12-22 10:00:12] Writing RTC time to system...
[2016-12-22 10:00:13] Done :-)
[2016-12-22 10:00:15] Pending for incoming shutdown command...
```

uugear, 2017-01-05:
When runScript.sh is running, it uses the current time in RTC to calculate the next startup/shutdown time. If it scheduled wrong, the error should come from the RTC time. However, after writing RTC time to system (on 2016-12-22 10:00:13), the system time is still correct. So I have no idea why it thought current date is 2016-12-23.

Is this issue reproducible?

gmh04, 2017-01-06:
Yes it is reproducible but not exactly as shown.  It varies on how quickly the problem occurs, sometimes after a day sometimes after a few days, and never at the same time.  I have put it wholly on battery power outside on 3rd Jan, so have yet to check how it performed.