uugear/Witty-Pi #29: Witty Pi 2 with Raspberry PI 3 model B does not start

mtrambaioli, 2017-03-15:
Hello
I have been trying to get the Witty Pi 2 working with Raspberry Pi 3 model B for couple of week but not able to make it function. I have followed the instructions but I get consistently the same behaviour, the PI3 shuts down if I press the button (scheduled shutdown does not work) but never restarts. I restart the PI3 by pressing the button, the red and green light starts, I can see the multicolor splash screen but then it shuts down. I have 2 Witty Pi2 and 3 PI3 so have tried with a fresh set and behaviour is the same. I have then one last time resinstalled everything:

fresh install of raspian
Installed wittypi software as per instructions
shut down PI3 and installed the Wittypi2 phisically
restarted Wittypi2 by plugging the power in witty pi2. Did not start (red and green light plus splash screen and then shuts down). so plugged in PI3
have configured the shutdown time and restart
I have been able to start it by connecting pin 1 and 4, this way it starts. I have also tried to use custom pin (6 and 8) but no difference. Can you please tell me what I'm doing wrong.
Thank you
Massimo

uugear, 2017-03-15:
When you said pin 1 and 4, do you mean the TX pin and +5V? If so please don't do this again, because all Raspberry Pi pins are in 3.3V level and they might not survive in 5V.

However, if you pull up the TX pin and it can start, then I believe you are encountering the issue mentioned here: [http://www.uugear.com/portfolio/is-witty-pi-compatible-with-raspberry-pi-3/]()

Although the article was written for Raspberry Pi 3, some customers do confirm it also apply to Raspberry Pi 2.

Please physically remove Witty Pi 2 from your Pi3 and then start it, run "gpio readall" and you will be able to confirm it.

mtrambaioli, 2017-03-15:
Hi
I have downloaded the latest software so don't believe this is the issue. I had already checked this article and checked config.txt has:
dtparam=i2c1=on
dtoverlay=pi3-miniuart-bt
core_freq=250

Regarding the pin 1 and 4 I mean to say physical pin 1( 3.3v) and GPIO 4 (physical pin 7). I followed example foung here: http://www.uugear.com/portfolio/use-witty-pi-2-to-build-solar-powered-time-lapse-camera/

mtrambaioli, 2017-03-15:
What would help is a step by step instruction on how to install on the Pi3 model B . I can start from scratch and report where it does not behave as expected. As mentioned I have 3 PI3 and 2 Witty Pi2 and I get same behaviour regardless of which ones I use

uugear, 2017-03-15:
The user manual has detailed description about how to install and use Witty Pi 2. I believe you are encountering something special. Please, try to boot your Raspberry Pi 3 and run "gpio readall" command, and then paste the output here.

Also, please let me know how do you install the OS, do you use NOOBS? Or you just flash the SD card directly.

mtrambaioli, 2017-03-15:
i have followed the Witty Pi 2 manual. Raspian is installed through NOOBS and updated with: sudo apt-get update and sudo apt-get upgrade 
Here is the output requested:
 pi@raspberrypi:~ $ gpio readall
 +-----+-----+---------+------+---+---Pi 3---+---+------+---------+-----+-----+
 | BCM | wPi |   Name  | Mode | V | Physical | V | Mode | Name    | wPi | BCM |
 +-----+-----+---------+------+---+----++----+---+------+---------+-----+-----+
 |     |     |    3.3v |      |   |  1 || 2  |   |      | 5v      |     |     |
 |   2 |   8 |   SDA.1 | ALT0 | 1 |  3 || 4  |   |      | 5v      |     |     |
 |   3 |   9 |   SCL.1 | ALT0 | 1 |  5 || 6  |   |      | 0v      |     |     |
 |   4 |   7 | GPIO. 7 |   IN | 1 |  7 || 8  | 1 | ALT0 | TxD     | 15  | 14  |
 |     |     |      0v |      |   |  9 || 10 | 1 | ALT0 | RxD     | 16  | 15  |
 |  17 |   0 | GPIO. 0 |   IN | 0 | 11 || 12 | 0 | IN   | GPIO. 1 | 1   | 18  |
 |  27 |   2 | GPIO. 2 |   IN | 0 | 13 || 14 |   |      | 0v      |     |     |
 |  22 |   3 | GPIO. 3 |   IN | 0 | 15 || 16 | 0 | IN   | GPIO. 4 | 4   | 23  |
 |     |     |    3.3v |      |   | 17 || 18 | 0 | IN   | GPIO. 5 | 5   | 24  |
 |  10 |  12 |    MOSI |   IN | 0 | 19 || 20 |   |      | 0v      |     |     |
 |   9 |  13 |    MISO |   IN | 0 | 21 || 22 | 0 | IN   | GPIO. 6 | 6   | 25  |
 |  11 |  14 |    SCLK |   IN | 0 | 23 || 24 | 1 | IN   | CE0     | 10  | 8   |
 |     |     |      0v |      |   | 25 || 26 | 1 | IN   | CE1     | 11  | 7   |
 |   0 |  30 |   SDA.0 |   IN | 1 | 27 || 28 | 1 | IN   | SCL.0   | 31  | 1   |
 |   5 |  21 | GPIO.21 |   IN | 1 | 29 || 30 |   |      | 0v      |     |     |
 |   6 |  22 | GPIO.22 |   IN | 1 | 31 || 32 | 0 | IN   | GPIO.26 | 26  | 12  |
 |  13 |  23 | GPIO.23 |   IN | 0 | 33 || 34 |   |      | 0v      |     |     |
 |  19 |  24 | GPIO.24 |   IN | 0 | 35 || 36 | 0 | IN   | GPIO.27 | 27  | 16  |
 |  26 |  25 | GPIO.25 |   IN | 0 | 37 || 38 | 0 | IN   | GPIO.28 | 28  | 20  |
 |     |     |      0v |      |   | 39 || 40 | 0 | IN   | GPIO.29 | 29  | 21  |
 +-----+-----+---------+------+---+----++----+---+------+---------+-----+-----+
 | BCM | wPi |   Name  | Mode | V | Physical | V | Mode | Name    | wPi | BCM |
 +-----+-----+---------+------+---+---Pi 3---+---+------+---------+-----+-----+
pi@raspberrypi:~ $

uugear, 2017-03-15:
If you install OS via NOOBS, the NOOBS boot menu will delay the booting for a few seconds and Witty Pi 2 will think your OS is not up and then cut the power.

Please follow the instructions here to skip the NOOBS boot menu: [http://raspberrypi.stackexchange.com/questions/34508/how-to-skip-noobs-boot-menu](http://raspberrypi.stackexchange.com/questions/34508/how-to-skip-noobs-boot-menu)

I believe it will solve the problem.

mtrambaioli, 2017-03-15:
this did the trick, now all working properly. thank you for your help

closed 2017-03-15 by uugear