uugear/Witty-Pi #35: Unrequested sleep

MihaMarkic, 2018-02-03:
I have a loop where I do some processing that instruct wittyPi to sleep and wake up after 4min (or something less). It mostly works, but occasionally raspberry will wake up correctly and go to sleep even before my program starts or in the middle.

Logs show something like this:
[2018-02-03 14:20:03] Done :-)
[2018-02-03 14:20:05] Pending for incoming shutdown command...
[2018-02-03 14:20:21] Shutdown command is received...

While sleep is still set at day 14:17:25.
What might trigger sleep in such case? I can't find the reason. It happens randomly.

MihaMarkic, 2018-02-03:
Reading some more, it could be an unstable GP4, is there a way to find out the source of sleep command?

uugear, 2018-02-05:
If you see "Shutdown command is received..." in the log, it means the GPIO-4 pin was pulled down at that moment. If you are sure nobody (hardware/software) was pulling it down, then it could be electromagnetic interference. If it is the case, you can consider connecting a 4.7K resistor between GPIO-4 and 3.3V pins, an external pull-up resistor should avoid this kind of interference.

MihaMarkic, 2018-02-05:
Thanks, I'll try now with the resistor and see how it goes.

MihaMarkic, 2018-02-05:
I've put a 4,99K resistor and now it won't sleep anymore. It is stuck in half-sleep (red light is on, rasbperry isn't responding nor it wakes up).

uugear, 2018-02-09:
Are you using Witty Pi, Witty Pi 2 or Witty Pi 2 Rev2?
If you are not sure you can check the links below:
[Witty Pi vs Witty Pi 2](http://www.uugear.com/news/witty-pi-2-the-new-witty-pi/)
[Witty Pi 2 vs Witty Pi 2 Rev2](http://www.uugear.com/news/new-revision-of-witty-pi-2/)
For different model the situation may vary.

MihaMarkic, 2018-02-10:
Witty Pi, I thought it's implicit as Pi 2 has another repo.

uugear, 2018-02-10:
You are right. But I have to ask because some people post question about Witty Pi 2 here (as you can see).
For Witty Pi the pull up resistor is harder to decide. The best way is to temporarily connect a 20K potentiometer between GPIO4 and 3.3V pins. Try to choose the smallest value that can still let Witty Pi cuts the power (red light goes off after shutdown), then replace the potentiometer with a fixed resistor.