uugear/Witty-Pi #36: Witty Pi 2 with Raspberry PI 3 model B does not start (no noobs)

Joehc, 2018-03-04:
I have successfully installed the software, and I have mounted the witty pi, and the white LED is fading in and out slowly, so I guess is ready to go. But whenever I press the power button, the raspberry pi is running in some seconds, but then it shuts down.

So I guess the problem is very similar if I had NOOBS installed, so the witty pi cant see my OS is up, but I did not use NOOBS.

What else can this be and how do I debug it?

bbillp, 2018-03-05:
Remove all power, remove the witty pi battery, wait a few minutes , add power and boot up, delete any schedule files then do sudo ./wittypi.sh.

uugear, 2018-03-07:
If you are using Witty Pi 2, you should post this issue to another repository (this repository is for the first generation of Witty Pi).

Basically there are 3 possibilities:
1. Serial port is not properly configured. You can run "gpio readall" to confirm the pins' status. [More details here](http://www.uugear.com/portfolio/is-witty-pi-compatible-with-raspberry-pi-3/)
2. The actual booting is delayed for a few seconds, and Witty Pi 2 cuts the power. If you have Raspberry Pi connected to TV, and you don't see Raspberry Pi logo before power cut, then it is the case.
3. GPIO-4 is pulled down (by software or hardware).

Joehc, 2018-03-10:
It was the serial port wich was not configured correct. Thx!