uugear/Witty-Pi #37: Next Up Time not Scheduled

mcsarge, 2018-03-21:
I have the Witty Pi Mini  and I have created a simple schedule that should wake the PI every 10 minutes if it happens to have been shut down. I assumed that if the Pi is up when the wake up time went by, then it would reschedule for the next wake up time. What I am seeing is that does not happen, the wake up time goes into the past and so is never performed. Is this correct behavior? Here is the schedule:

BEGIN 2018-03-01 00:00:00
END   2025-01-01 00:00:00
ON    M1 WAIT
OFF   M9

When I launch the schedule it correctly schedules the boot up for the next 10 minute mark, but if I do not shut down the Pi before that time, the scheduled boot up time does not get moved to the next 10 minute mark.

When I shut down the Pi I always us the "gpio mode 7 out" to shut it down. Am I doing this wrong?

uugear, 2018-03-21:
When using the "WAIT" syntax, you need to shutdown your Pi before the scheduled startup time, otherwise the schedule script can not continue. What you saw is expected behavior.

They way you shutdown the Pi is correct, you just need to do it before the next startup comes.

mcsarge, 2018-03-21:
OK,

I think this is wrong behavior, it is entirely possible that the Pi will not go down before the end of the UP time, and if that happens, it will never come back up again. I think that the wittyPi system should calculate and update the next wake time before it shuts down the Pi. It would be simple to do in the daemon script. I do not believe this would cause any issue with other running scripts and makes the function more predictable. I will make a change in the daemon and test and provide it if you want.

uugear, 2018-03-21:
Usually we use "WAIT" syntax to shutdown the Pi earlier than the scheduled shutdown, to reduce the power consumption. The typical use case is time-lapse camera: the Pi wakes up, takes a picture and turns off immediately. We have an example [here](http://www.uugear.com/portfolio/use-witty-pi-2-to-build-solar-powered-time-lapse-camera/).

Didn't think of the need to do the other way around. Are you trying to use it like a watchdog, to make sure the system keeps being turned on?

There is no guarentee that daemon.sh has chance to do something before the Pi being shutdown, considering the possibility of accidently losing power, or the user run "sudo shutdown now" instead of "gpio mode 7 out". However if you want to keep adjusting the scheduled startup time, maybe the best bet is to invoke the "runScript.sh" at line 96 in daemon.sh, where the script has detected the startup alarm occurs while system is still on.

mcsarge, 2018-03-22:
Sort of. I have a system I am putting together that will monitor it's solar power and shut down gracefully if it gets too low, But I need it to wake up now and then to check the power to see if has improved. I will keep thinking about it. Right now, I just run the runScript.sh every minute to update the next wakeup time, which seems excessive.