uugear/Witty-Pi #38: Witty Pi across the world

Taramatalchum, 2018-03-28:
Hi !

I am preparing an object that contains a Raspberry Pi 3 B, with a Witty Pi attached, and a screen. It displays a looped video all day.

This object will be shipped to the USA for a show, and will then travel to Japan and Korea.

It is programmed to turn on and off at certain hours, defined by the art galleries. They also have to give wifi info so that i can enter it in advance in wpa_supplicant.conf. This way, the object will (hopefully) be connected.  There will be an ethernet plug if wifi doesn't work. I also have installed Dataplicity so that I can control the object from Europe, where I'm based.

My question is : if the object travels around the world without coming back to me, will I be able to reprogram it on the fly so that it adapts to the different time zones and wifi country codes and on/off schedules ?

The Witty Pi is working fine here, but when I tried changing time zone/wifi country code AFTER having installed the Witty Pi, I had some issues (time off by 1 hour even with internet connection, or the Witty Pi adding one hour to the programmed on/off schedule (when I entered ?? 17:00 for example, it programmed ?? 18:00)).

Is the Witty Pi reliable for such a trip ? How can I make it 100% reliable ?

Tell me if you need more info.

Thanks in advance !

P.S.: I'm a total newbie. I'm learning as I go, so excuse me if I'm not doing this right.

uugear, 2018-03-28:
If you are using Witty Pi 2, you should submit this issue to the repository for Witty Pi 2. This repository is for the first generation of Witty Pi (retired).

As long as your device (object) has Internet connection and you know its IP address, you can use SSH to remotely login and make any change you need.

As for the "adding one hour" issue, it is a known display bug, and it only happens when your country is going start daylight saving time in current month, and you are using "??" wildcard in the date field. It would not actually affect the schedule you defined, it is just a display bug. Also you will not see this bug in April.

To avoid confusing Witty Pi's RTC, you should remotely login to your Raspberry Pi and configure the timezone and activate the schedule script again, to make sure the scheduled shutdown/startup time is correct.

Taramatalchum, 2018-03-28:
Thanks for the quick answer !
It's good to know it's only a display bug. So I hope the travel will go fine !

And thanks for the Witty Pi. It's a very great tool.